porrey/max1704x #1: Question about power consumption

MoJo2600, 2019-06-05:
Hi,

this is not an issue in your library, but maybe you are able to help me, because you have more experience with the MAX17043.

I'm using an nodemcu esp-12F and without the MAX17043 connected the power consumption is around 0.01 mA in deep sleep. I connected the Fuel Gauge in parallel to my battery and I receive readings from the gauge and everything works well. Except, that the power draw is very high. In deep sleep of the esp8266 the current that is being drawn is around 0.9mA. When i disconnect the positive battery wire from the fuel gauge, the power consumption drops to around 0,05 mA. This would match my expectation, as the MAX17043 has an active current from 0.05 mA to 0.075 mA - Putting the MAX17043 device to sleep has no effect. Fun fact, if i reconnect the power lead to the MAX17043, the power consumptions stays at 0.05 mA. I have no idea what could be causing this.
I tried differend nodemcu boards, but all show the same behaviour. Do you have an idea, why the power consumption is so high?

Best regards

christian.2.erhardt@mojo2k.de

porrey, 2019-06-05:
Can you provide more details?

1) Are you using a break-out board such as the LiPo Fuel Gauge from SparkFun?
2) Can you send a link to your node-mcu board so i can get more info on it?
3) Are you using a LiPo charger as well? If yes, which one? 
4) If you can send details of your circuit (diagram or photos) that would be helpful.

MoJo2600, 2019-06-06:
Thank you, that you are willing to have a look.

1. Yes, i use the LiPo Fuel Gauge from Sparkfun
2. The NodeMCU i'm currently using is a ESP-12F in detail it is this soil moisture sensor i try to modify to use lipos instead of batteries (https://wiki.aprbrother.com/en/ESP_Soil_Moisture_Sensor.html) but i observed same behavior with an Sparkfun ESP8266 thing
3. Yes, i use one the TP4056 based charging circuits with protection - the nodemcu is powered by a mcp1700 ldo regulator
4. Yes i can create a picture when i'm at home. But the circuit is very straight forward. The charging circuit and the LiPo fuel gauge are connected in parallel to the battery. Battery out of the TP4056 is connected to the mcp1700. The connection to the nodemcu is only SDA/SCL and GND. 

Without the LiPo Fuel Gauge connected, the current reading goes down to about 0.02 mA when the ESP is going to deep sleep. When I connect the Fuel Gauge, the current stays at 0.9 mA. When i disconnect the battery lead from the fuel gauge, the reading drops to 0.05 mA and stays there even if i reconnect. 

I think there is some side effect between mcu / charger and FuelGauge but i have no idea what it could be.

Thank you!

MoJo2600, 2019-06-06:
Hm... i removed the charging circuit, same effect. Still 0.9 mA. But if i disconnect the NodeMCU, the voltage drops to 0.05 mA which would match the spec of the MAX17043. So what on the NodeMCU side could cause this waste.

porrey, 2019-06-06:
Do you have a plain/ordinary Arduino such as an Uno or another non-MCU device to try? Also, are you sure it is going into deep sleep mode? Perhaps one of the connections is causing it to fail to go into deep sleep mode. 

I'll keep digging but nothing comes to mind right now.

MoJo2600, 2019-06-10:
Ok, i think i figured it out. Maybe you're interessted in the result. The Lipo Fuel Gauge has pull up resistors on the i2c bus, directly connected to the battery input. As my NodeMCU ESP8266 is running on  3.3 V and the battery is delivering up to 4.1 V there was a small voltage potential that draws some current. I cut the trace between the battery and the VCC of the MAX chip and connected it to the 3.3 V of the NodeMCU. This dropped the current draw from 0.9 ma to 0.75 mA. Better but not satisfying.
I diged a little deeper and read, that i2c should only have one set of pull-up resistors on the SDA and SCL line. The ESP8266 seems to use the internal pull-up resistors on the SDA and SCL line. I disabled them and success, the current draw dropped from 0.75 mA to 0.08 mA and stays there.
I disabled the internal pull up resistors with this few lines of code

```
// Skip fuelgauge.begin(), as it calls Wire.begin(), which we call
// fuelgauge.begin()
Wire.begin();
digitalWrite(SDA, 0);
digitalWrite(SCL, 0);
```

closed 2019-06-10 by MoJo2600