porrey/max1704x #4: Library should not initialize I2C subsystem itself

petrkr, 2021-04-25:
I do not know why most of Arudino libraries trying do something what they should not do.

Here you initialize i2c bus. That is problem if you want to use I2C on different pins (you cannot pass begin(sda,scl). Also if you have bus initialized from outside, you reinitialize it by own and break other I2C libraries

I suggest to add another constructor with I2C reference class, so it will NOT break up old behaviour and also it introduce way how to reuse initialised I2C interface from code.

https://github.com/porrey/max1704x/blob/99644a0b7d4a83bbb409b319e885dc52b3cf5a1d/src/MAX1704X.cpp#L33

porrey, 2021-04-25:
I agree and will add a new constructor.

Note as workaround that **`begin()`** does not need to be called on this library since it only initializes the i2c bus. Not calling this constructor and calling **`Wire.begin(sda,scl)`** will allow the library to work on other pins.

porrey, 2021-05-03:
Updated library with new constructors. See v1.1.0.

closed 2021-05-03 by porrey