porrey/max1704x #5: Inconsistent values

domix75, 2021-05-01:
Hello

I just received this and try to make it work, it's just a test settings with a nano and the fuel jauge
got it connected and uploaded your library 
the result seems inconsistent :

20:19:03.158 -> Version:   3
20:19:03.158 -> ADC:   3923
20:19:03.158 -> Voltage:   4903.75 v
20:19:03.158 -> Percent:   190.84%
20:19:03.158 -> Is Sleeping:   No
20:19:03.158 -> Alert: No
20:19:03.158 -> Threshold: 4
20:19:03.158 -> Compensation:  0x97

I measure the battery with a controler => 4.2V (it's fully charged, it's a lipo 3.7 250mAh cf. https://www.gotronic.fr/pj2-09941-1983.pdf)

the result is identical if the battery is there or not

apparently it recognises it since when i disconnect the jauge i get 
0:24:29.086 -> Version:   65535
20:24:29.086 -> ADC:   4095
20:24:29.086 -> Voltage:   5118.75 v
20:24:29.086 -> Percent:   256.00%
20:24:29.086 -> Is Sleeping:   Yes
20:24:29.086 -> Alert: Yes
20:24:29.086 -> Threshold: 1
20:24:29.086 -> Compensation:  0xFF

any idea where I could search ?
thanks
dom

porrey, 2021-05-03:
First things first, I noticed the example code display the voltage as V instead of mV. The output of **`FuelGauge.voltage()`** is in **mV** and not V (my mistake). In your example, the reading 4903.75 v is really 4.9 V.

I connected my Spark Fun fuel gauge to a Nano with a li-po battery measuring about 3.9 v. The fuel gauge out of the example is:

```
Version:       3
ADC:           3178
Voltage:       3972.50 mV
Percent:       72.41%
Is Sleeping:   No
Alert:         No
Threshold:     4
Compensation:  0x97
```
(I changed v to mV).

I connected the GND pin to GND on the Nano, the SDA pin to A4 and the SCL to A5. I left the Vcc,  pin on the fuel gauge disconnected. With Vcc connected to the 5V pin on the nano I get the same result. Note that leaving Vcc disconnected allows the battery to drain.

I tested a handful of other batteries and all reported proper voltage and percentage correctly.

What version of the breakout board do you have? I ask because I was able to produce similar results using a (random) different address. I read that some boards use **0x32** (the library uses 0x36). See the discussion in this **[Forum](https://forum.arduino.cc/t/max17043-fuel-gauge-not-working-at-all/329534/17)**.

I will be releasing an updated library that allows specifying the address. In the meantime you can try copying the library files into the folder where your INO file is and change the address in **MAX1740X.H** to **0x32**.

domix75, 2021-05-03:
Thank you for your answer,
OK I understand for the mV/V
it seems mine is a genuine SparkFun, it's on a red board with their logo (kinda flame, in white) under the white text "Fuel Jauge", the connector is black. 
besides I checked with a I2C scanner and it's on the 0x36

I'll do some more checkings tonight without the Vcc. 
I just connect SDA, SCL and GND, directly on the corresponding pins, without any resistor

cheers

porrey, 2021-05-03:
Also try connection the QSTRT pin to GND.

domix75, 2021-05-03:
I eventually had it work.

There are 2 issues but I'm not sure the hardware can do better, are measuring a jauge is quite hard...
- one is the value of "percent" : the lipo is sold for 3.7 and the specif says that it's max is 4.2, then when I calculate the max value I should have according to the voltage and the percent I endup by 4.5/4.6
- the other issue is that I was connecting the VCC. this was making the percent become totally inconsistent (like 3% instead of 87% and a few min later, showing the same voltage, it was 45...)

as for the QSTRT it does not change much the readings in my case

thanks again and have a great day
dom

porrey, 2021-05-04:
I'm glad you sorted it out. Here is some final information as I close the issue that may help others reading this exchange.

I checked a few other batteries that I had, particularly some "cheap" ones and found that they measure over 100%. The voltage on these was higher than expected. Most of the batteries I use I have obtained from [Adafruit](https://www.adafruit.com/category/917) or [SparkFun](https://www.sparkfun.com/search/results?term=battery). These batteries seem to be the highest quality and work the best.

Here is a picture of my circuit if it helps at all. Note the reading from the output of the Fuel Gauge is is **3.76 V**. I checked this with a Fluke 87 and it is spot on. The out put from the program is:

```
Device Reading:
Address:       0x36
Version:       3
ADC:           3010
Voltage:       3762.50 mV
Percent:       27.82%
Is Sleeping:   No
Alert:         Yes
Threshold:     32%
Compensation:  0x97
```

[image]

[image]

[image]

[image]

closed 2021-05-04 by porrey

domix75, 2021-05-04:
Thanks 

well maybe there is somth. I had missunderstood in the function of the battery. I see on yr example that the volatge is 3.76 and the percent is 27% so the relation betw voltage is more complicated that I had understood. 
with the values I got on https://blog.ampow.com/lipo-voltage-chart/ my readings are actually consistent.