porrey/max1704x #10: Clone max1743 address 0x76 wrong readings

Darozas, 2021-08-20:
Hi 
Is there a way to calibrate readings got clone chip that reads address as 0x76 but getting wrong readings.
Baterry is charged to 3.8v but reads as 2.2v and capacity is 0

Can you help here please

porrey, 2021-08-20:
Can you provide some information about how you are connected? What type of microcontroller are you using and at what voltage?

Darozas, 2021-08-21:
Hi 
I have connected as per below.
Controller is Esp32. I get same results with Uno aswell

Thanks
[image]

cranefist, 2021-08-24:
I have one that has 0x32 address, and it seem to only work by first putting it into sleep, waking up and then reading it in fast succession. The percentage is still reported incorrectly.

Darozas, 2021-08-24:
Hi 
Same here i have purchased one from ebay and its came on 0x32 and shows me 4% while battery still on 3.7v.
I dont see how you can 18650 to go to 3v.

cranefist, 2021-08-24:
> Hi
> Same here i have purchased one from ebay and its came on 0x32 and shows me 4% while battery still on 3.7v.
> I dont see how you can 18650 to go to 3v.

I opened a separate issue for the 0x32 model, check that out. It has a way to make it work, it would need to be just implemented better.

porrey, 2021-08-24:
Please see https://forum.arduino.cc/t/max17043-fuel-gauge-not-working-at-all/329534. There are a lot of articles on the Internet about these clones being unstable. Some report them working properly while most say hey are not consistent. I am open to adjusting the library to work with these clones but I am not sure what can be done.

porrey, 2021-09-12:
I picked up a clone that works on address 0x32. Communication works well but the device does not measure voltage (VCELL)  or SOC correctly. The VCELL registers 0v on every call.  Calling quick start and then taking  a reading gives a voltage reading other than 0, but neither this reading nor the SOC is accurate. Other functions seem to work as usual, but these clones chips do not operate per the MAX1704X data sheet. This behavior is consistent with what I see other people experiencing with these chips. There is no issue with this library and therefore I am closing this issue.

closed 2021-09-12 by porrey