porrey/max1704x #11: Aliexpress 0x32 Clone Support

cranefist, 2021-08-24:
Would it be possible to support the Aliexpress clone boards that have 0x32 address? They seem to work, but work differently.

There is one post here, that shows how it works:
https://forum.arduino.cc/t/max17043-fuel-gauge-not-working-at-all/329534/15

I tested this, by rapidly putting it to sleep, waking it up and then reading. This way it gets a reading, other than that it seems to not read the voltage. The percentage is still wrong though.

[image]

porrey, 2021-08-24:
I am willing to make and changes necessary and provide support for clones. Do you have any idea what is needed? I need to try to purchase at least one clone to experiment with.

cranefist, 2021-08-25:
Currently, all i got is how to read it... i don't understand the readings yet, they do work in the sense that the readings go down as the battery is used. Just not sure how to get them right.

The original chip has a measuring range upto 5v, but this seems to top out at 4.2v. Because a fully charged and measured at 4.2v cell gives you the output of 4.9v. So maybe the 10bit readout is just scaled for the max of 4.2v.

Percentages go up to around 200% at fully charged. So there is some logic behind this, just need to figure it out.

cranefist, 2021-08-28:
I hooked up to my bench power supply. It tops out at 4.3v like it though. 4.364v according to my multimeter, at this point the chip reads 5115mv... and if you go over that, it loops back to 5mv. So the 10bit readout is scaled to 0 - 4.3v. Instead of the 0-5v.

At 4.25v it reads 200%. And at 2.75v its at 0%. These are both multimeter readings.

cranefist, 2021-08-28:
I just divided the percentage by 2, and i think that fixes that.. then the 10bit voltage readout needs to be just correctly scaled. And that should fix that.

cranefist, 2021-08-28:
Adjusted the increments to 1.06 in the MAX17043.h. Now the voltage readings are pretty much accurate.

cranefist, 2021-08-28:
`#include "MAX17043.h"

void setup()
{
  
  Serial.begin(115200);
  while (!Serial) {}
  Serial.println("Serial port initialized.\n");
  
  FuelGauge.begin();

}

void loop()
{

  FuelGauge.quickstart();
  delay(100);
  displayReading();
  delay(500);
 
}

void displayReading()
{
  Serial.print("Voltage:       "); Serial.print(FuelGauge.voltage()); Serial.println(" mV");
  Serial.print("Percent:       "); Serial.print(FuelGauge.percent()/2); Serial.println("%");
  Serial.println();
}`

cranefist, 2021-08-28:
[image]

Well, if this is what the original outputs.. then the percentages are still way off.

porrey, 2021-08-29:
Can you try changing the compensation value to see how or if it affects the readings?

cranefist, 2021-08-30:
> Can you try changing the compensation value to see how or if it affects the readings?

Already did, that's what fixed the voltage readings.

The issue now is the percentage readings, they are not correct at all.

porrey, 2021-09-05:
I have a clone on order and will start to test with it as soon as I get it.

porrey, 2021-09-12:
I received my clone and did some testing side by side with a SparkFun chip. The readings below are from the same battery.

SparkFun Chip:
```
Device Reading:
Address:       0x36
Version:       3
ADC:           3230
Voltage:       4037.50 mV
Percent:       80.59%
Is Sleeping:   No
Alert:         No
Threshold:     4%
Compensation:  0x97
```

Clone:
```
Device Reading:
Address:       0x32
Version:       3
ADC:           3772
Voltage:       4715.00 mV
Percent:       166.95%
Is Sleeping:   No
Alert:         Yes
Threshold:     4%
Compensation:  0x97
```

cranefist, 2021-09-20:
Did you change the voltage increment for this?

porrey, 2021-09-20:
In the above example, the increments can be changed to something like 1.07 to match but the reading wasn't consistent. There is also the issue that it always returns 0 unless quick start is used just prior to the VCELL reading. Lastly, even if the VCELL reading is compensated, the percentage is way off and there is no way to adjust this. The internal formula is not linear so it wold require knowing how the percentage is calculated to somehow applying a "smart" adjustment. All of this seems more work than needed to get these chips working.

If you can share the files/information you received I can spend some time looking into it.

cranefist, 2021-09-25:
Yes, you always have to do the quick start before the reading. And the correct adjustment was 1.06, and that makes sense considering how it tops at 4.3v instead of 5v having 12bit range.

The percentage was the actual issue, as that does not make sense.

github0013, 2021-12-24:
Just FYI, but I [bought this one][1] (I believe it's a copy of Sparkfun's?), and [the address was 0x36][2].
I used your library (porrey/max1704x), and it worked just fine without modifying the code.

[1]: https://www.aliexpress.com/item/4000007759641.html
[2]: https://randomnerdtutorials.com/esp32-i2c-communication-arduino-ide/#2

porrey, 2021-12-24:
> Just FYI, but I [bought this one](https://www.aliexpress.com/item/4000007759641.html) (I believe it's a copy of Sparkfun's?), and [the address was 0x36](https://randomnerdtutorials.com/esp32-i2c-communication-arduino-ide/#2).
> I used your library (porrey/max1704x), and it worked just fine without modifying the code.

This one appears to have a genuine Maxim Integrated chip instead of a clone. The price is good too.

porrey, 2021-12-25:
Clones of the MAX1704X chip appear to not work accurately. I have found no way to adjust this library to work with them. If anyone stumbles across a way, I will be more than happy to update the library as log as it does not affect the genuine chips. For no I am closing this issue as there are no apparent issues with the current library.

closed 2021-12-25 by porrey