porrey/max1704x #13: Stopping charge

domix75, 2022-01-17:
Hello and happy year 2022 

This is not an issue about your lib but a question. Do you have an idea of how to stop the charge when the fuel jauge says it's over ninety-ish or a bit more. I'm using the Max 1703 fuel jauge plus a lipo charger (https://www.dfrobot.com/product-2068.html). 
thanks, regards
dom

porrey, 2022-01-17:
**OPTION 1**: The board you are using has a TP4056X chip. Pin 8 on this chip is an _**Enable**_ chip (CE). A `HIGH` signal enables the chip. Pulling the pin `LOW` can disable the chip. There does not appear to be any breakout for this pin but perhaps you could solder a wire to it. **CAUTION:** I did not see a schematic for this board. If the pin is floating (not connected to anything) this option will work. If the pin is connected to the Vcc, then grounding this pin will short out the power supply. It appears the CE pin may be internally pulled `HIGH` which means it can be safely connected to GND but I cannot be 100% sure.

**OPTION 2:** Use an N-channel MOSFET to turn the power input on and off. This would require the ability to cut the negative connection of the input voltage which means you would not be able to use the USB port. Connect the gate of the MOSFET (G) to the I/O pin on your microcontroller, the Source pin (S) to GND and then connect the Drain pin (D) to the negative input pin on the LiPo Charging board. Connect the positive input pin of the LiPo charging board to +5V. A `LOW` signal to the G pin on the MOSFET turns the board off while a `HIGH` signal will turn the board on. Use something like this or equivalent (https://www.adafruit.com/product/355).

domix75, 2022-01-17:
Thanks a lot for your quick and precise response. About the option 1, probably I can check this with a multimeter. though I'm not sure I'll be able to solder on the tiny pin. I was thinking of something like option 2 but of course there is the pbm of powering the chip itself. 
Could I use the mosfet to cut the connexion from the charger to the battery though ?

porrey, 2022-01-18:
When I outlined option one I was assuming the battery could supply power, however, you can cut the battery connection as well. The MOSFET is a switch between the Source and Drain that is opened and close using the Gate. 

The **Source** on the MOSFET will stay connected to ground. Connect the negative battery terminal of your charger to ground. Then connect the **Drain** to the negative side of the battery. Connect the positive terminal of the battery to the positive terminal on the charger board. The MOSFET will connect/disconnect the battery to/from ground. When the **Gate** pin is `HIGH`, the battery is connected to ground and in turn the charger. When the Gate is `LOW`, it is disconnected.

Mind you, I haven't actually tested this...

domix75, 2022-01-21:
Thanks ! Your feedback is very useful to me

domix75, 2022-02-08:
Hello
unfortunately I can't make this work. 
I use a N-Type MOSFET (IRFZ24NPBF), for test purpose I put a button to switch the MOFSET gate from HIGH to LOW and I can see it works with a led. But When I replace the led by the battery, whether the gate is HIGH or LOW, the battery does not seem to charge (I check with a USB controler on the charger : it stays around 3mA, when charging pulls ~200-250mA - that I can see when I connect the ground of the battery directly on the ground...)

porrey, 2022-02-13:
I will work on a similar circuit and provide the details if I get it working.

domix75, 2022-02-13:
that's very kind, thanks

porrey, 2022-03-28:
Sorry it so long to respond but got real busy...

I was able to get the N-Channel MOSFET to work but found the doing so interfered with the microcontrollers ability to read the MAX17043. Here is an image of the circuit with an N-Channel MOSFET.

[image]

I switched to a PNP MOSFET to connect and disconnect the power to the charger and this worked perfectly. Here are images of the circuit with an P-Channel MOSFET.

The circuit is charging (0V to the gate)
[image]

The circuit is not charging (5V to the gate)
[image]

_The charger I am using has a buck converter that regulates the battery to 5V when the battery is not charging._

Here is a crude circuit diagram using the P-Channel MOSFET:
[image]

_Note I just used a loose green wire to turn the MOSFET on and off. It does not need to be connected, just momentarily touch either GND or 5V._

closed 2022-10-21 by porrey