porrey/max1704x #15: I'm having issues regarding showing the battery percentage.

Cicrumstonce, 2022-02-23:
So, for the last month I've been using this code and the module that someone suggested to and everything in your code works, except one thing that's quite crucial to my project. Which is showing the percentage of the battery through my LCD. I've made sure everything was connected properly and everything else worked properly which it does fortunately. I can show the voltage on the LCD but the percentage constantly stays at 0.00% which is rather strange to me.
[image]

You'll see that I have a voltage going through, but no percentage. I've tested the other properties in your code and they work properly.

porrey, 2022-02-24:
I added a standalone sketch called Debug to the examples folder. It will display the device register values as well as the calculated values. This may help debug your issue.

Output from SparkFun fuel gauge:

```
Serial port initialized.

ADDRESS...........: 0x36	(54)
VCELL HIGH BYTE...: 0xC3	(195)
VCELL LOW BYTE....: 0x40	(64)
SOC HIGH BYTE.....: 0x40	(64)
SOC LOW BYTE......: 0xB3	(179)
MODE LOW BYTE.....: 0x00	(0)
MODE LOW BYTE.....: 0x00	(0)
VERSION LOW BYTE..: 0x00	(0)
VERSION LOW BYTE..: 0x03	(3)
CONFIG LOW BYTE...: 0x97	(151)
CONFIG LOW BYTE...: 0x1C	(28)

PERCENT...........: 64.70%
ADC...............: 3124
VOLTAGE...........: 3905.00 mV
```

Output from clone device (this device does not work):

```
Serial port initialized.

ADDRESS...........: 0x32	(50)
VCELL HIGH BYTE...: 0xFA	(250)
VCELL LOW BYTE....: 0xC0	(192)
SOC HIGH BYTE.....: 0xCC	(204)
SOC LOW BYTE......: 0xEC	(236)
MODE LOW BYTE.....: 0x00	(0)
MODE LOW BYTE.....: 0x00	(0)
VERSION LOW BYTE..: 0x00	(0)
VERSION LOW BYTE..: 0x03	(3)
CONFIG LOW BYTE...: 0x97	(151)
CONFIG LOW BYTE...: 0x1C	(28)

PERCENT...........: 204.92%
ADC...............: 4012
VOLTAGE...........: 5015.00 mV
```

Cicrumstonce, 2022-02-25:
Alright, so from what I am seeing in the Serial monitor is that I am not receiving SOC HIGH and LOW, as well as VCELL LOW BYTE. 

```
Serial port initialized.

ADDRESS...........: 0x36	(54)
VCELL HIGH BYTE...: 0x86	(134)
VCELL LOW BYTE....: 0x00	(0)
SOC HIGH BYTE.....: 0x00	(0)
SOC LOW BYTE......: 0x00	(0)
MODE LOW BYTE.....: 0x00	(0)
MODE LOW BYTE.....: 0x00	(0)
VERSION LOW BYTE..: 0x00	(0)
VERSION LOW BYTE..: 0x03	(3)
CONFIG LOW BYTE...: 0x97	(151)
CONFIG LOW BYTE...: 0x7C	(124)

PERCENT...........: 0.00%
ADC...............: 2144
VOLTAGE...........: 2680.00 mV
```

porrey, 2022-02-25:
If the voltage is correct, the battery is low and 0% makes sense. The voltage should be above 3.7v to operate effectively. At 100% it is usually at 4.2v or more. It is odd, however, that the VCELL LOW byte is 0.

porrey, 2022-03-27:
The MAX17043 can only read a voltage of a single cell up to approximately 4.7V. Using two batteries in parallel would not be supported either.

Try issuing a reset() and then a quickstart() during initialization. See this example for more details: https://porrey.github.io/max1704x/examples/basic.

closed 2022-10-21 by porrey