porrey/max1704x #16: Battery never shows more than 90% charge

cservidoctavian, 2022-02-27:
Hello and thanks for all your work! 

I want to open a discussion on lipo battery that I am using with MAX17043. 

The battery never charges to 4.20, realistically it will charge to 4.15 or 4.18 and that means
the MAX17043 will never say the battery is 100% but it actually is because lets be honest 
0.05-0.010v means nothing really. 

Is pretty annoying because I am using it in my project and when it powers on,
voltage will drop FOR SURE even if is 4.20v. (~200mA current draw, with 2000mAh lipo battery)

**Solution:**

For now I think I will just ignore the percentage reading from the MAX17043 itself, and 
I will just implement an if range statement that will only look at the voltage and based 
on that I will generate percentage.

Can we fix this issue? I can help out on testing if that helps..
I think is a real issue because users will never see a battery with more than 90%

porrey, 2022-02-27:
What is the enhancement you are requesting?

cservidoctavian, 2022-02-27:
Create a side logic to handle the batteries that charge only up to ~4.10v to be seen as 100%

porrey, 2022-04-29:
You could try the map() function. 

https://www.arduino.cc/reference/en/language/functions/math/map/

BhargavVadaliya, 2022-05-02:
> You could try the map() function.
> 
> https://www.arduino.cc/reference/en/language/functions/math/map/

Okay. Thanks for the help.

closed 2022-10-21 by porrey