porrey/max1704x #19: ESP32 WEMOS LOLIN32 lite

tmargot, 2022-03-27:
Hi,
I tried the basic example and it runs perfectly, just reports percentage as 256% presumably because I have the FuelGauge connected to non-default pins. So been trying to initialize my own Wire.h specifying the required pins but I get the following panic error:

`19:24:38.226 -> A2      : 0x00000000  A3      : 0x00000036  A4      : 0x00000000  A5      : 0x00000100  
19:24:38.319 -> A6      : 0x3ffb8a28  A7      : 0x00000000  A8      : 0x80088830  A9      : 0x3ffb2720  
19:24:38.413 -> A10     : 0x3ffc18c0  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x00000001  
19:24:38.460 -> A14     : 0x00060f20  A15     : 0x00000001  SAR     : 0x0000001c  EXCCAUSE: 0x0000001c  
19:24:38.600 -> EXCVADDR: 0x00000134  LBEG    : 0x4008649d  LEND    : 0x400864ad  LCOUNT  : 0xfffffffa  
19:24:38.694 -> 
19:24:38.694 -> 
19:24:38.694 -> Backtrace:0x400d13a3:0x3ffb27600x400d1508:0x3ffb2780 0x400d1128:0x3ffb27a0 0x400d1148:0x3ffb27c0 0x400d1185:0x3ffb27e0 0x400d10e6:0x3ffb2800 0x400d2309:0x3ffb2820 
19:24:38.836 -> 
19:24:38.836 -> 
19:24:38.836 -> 
19:24:38.836 -> 
19:24:38.883 -> ELF file SHA256: 0000000000000000
19:24:38.929 -> 
19:24:38.929 -> Rebooting...
19:24:38.929 -> 9ZO"⸮P
19:24:38.929 -> YP[4L⸮%p⸮Battery percentage is Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
`

**My code is:**

`#include <MAX17043.h>
const int pinSDA = 22;
const int pinSCL = 19;

void setup()
{
  Serial.begin(9600);
  Wire.begin(pinSDA, pinSCL);
  FuelGauge.begin(false);
}

void loop()
{
  Serial.print("Battery percentage is ");
  Serial.println(FuelGauge.percent());
}` 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

tmargot, 2022-03-27:
I changed my code to: `FuelGauge.begin(&Wire, true, 0x32);` now I have a battery percentage and voltage of 0.

porrey, 2022-03-27:
try a reset() and and quickstart() as shown in this example: https://porrey.github.io/max1704x/examples/basic

tmargot, 2022-03-28:
Thanks for the reply, I have run the example suggested. First log has very high mV and percentage, then mV goes instantly to 0mV and the percentage slowly goes down to 0%. Any ideas? 

18:44:57.253 -> 
18:44:57.253 -> Wire.begin()
18:44:57.301 -> Resetting device...
18:44:57.532 -> Initiating quickstart mode...
18:44:57.671 -> Reading device...
18:44:57.671 -> 
18:44:57.671 -> Device Reading:
18:44:57.671 -> Address:       0x32
18:44:57.671 -> Version:       3
18:44:57.671 -> ADC:           3884
18:44:57.671 -> Voltage:       4855.00 mV
18:44:57.671 -> Percent:       184.67%
18:44:57.671 -> Is Sleeping:   No
18:44:57.671 -> Alert:         No
18:44:57.671 -> Threshold:     4%
18:44:57.671 -> Compensation:  0x97
18:44:57.671 -> 
18:44:57.671 -> Device Reading:
18:44:57.671 -> Address:       0x32
18:44:57.671 -> Version:       3
18:44:57.671 -> ADC:           3884
18:44:57.671 -> Voltage:       4855.00 mV
18:44:57.671 -> Percent:       184.67%
18:44:57.671 -> Is Sleeping:   No
18:44:57.671 -> Alert:         No
18:44:57.671 -> Threshold:     4%
18:44:57.671 -> Compensation:  0x97
18:44:57.671 -> 
18:44:59.682 -> Device Reading:
18:44:59.682 -> Address:       0x32
18:44:59.682 -> Version:       3
18:44:59.682 -> ADC:           0
18:44:59.682 -> Voltage:       0.00 mV
18:44:59.682 -> Percent:       183.72%
18:44:59.682 -> Is Sleeping:   No
18:44:59.682 -> Alert:         No
18:44:59.682 -> Threshold:     4%
18:44:59.682 -> Compensation:  0x97
18:44:59.682 -> 
...
18:49:07.535 -> Device Reading:
18:49:07.535 -> Address:       0x32
18:49:07.535 -> Version:       3
18:49:07.535 -> ADC:           0
18:49:07.535 -> Voltage:       0.00 mV
18:49:07.535 -> Percent:       71.43%
18:49:07.535 -> Is Sleeping:   No
18:49:07.535 -> Alert:         No
18:49:07.535 -> Threshold:     4%
18:49:07.535 -> Compensation:  0x97
18:49:07.535 -> 
18:49:09.536 -> Device Reading:
18:49:09.536 -> Address:       0x32
18:49:09.536 -> Version:       3
18:49:09.536 -> ADC:           0
18:49:09.536 -> Voltage:       0.00 mV
18:49:09.536 -> Percent:       70.61%
18:49:09.536 -> Is Sleeping:   No
18:49:09.536 -> Alert:         No
18:49:09.536 -> Threshold:     4%
18:49:09.536 -> Compensation:  0x97
18:49:09.536 -> 
18:49:11.537 -> Device Reading:
18:49:11.537 -> Address:       0x32
18:49:11.537 -> Version:       3
18:49:11.537 -> ADC:           0
18:49:11.537 -> Voltage:       0.00 mV
18:49:11.537 -> Percent:       69.62%
18:49:11.537 -> Is Sleeping:   No
18:49:11.584 -> Alert:         No
18:49:11.584 -> Threshold:     4%
18:49:11.584 -> Compensation:  0x97
18:49:11.584 -> 
...
18:51:58.775 -> Device Reading:
18:51:58.775 -> Address:       0x32
18:51:58.775 -> Version:       3
18:51:58.775 -> ADC:           0
18:51:58.822 -> Voltage:       0.00 mV
18:51:58.822 -> Percent:       1.85%
18:51:58.822 -> Is Sleeping:   No
18:51:58.822 -> Alert:         Yes
18:51:58.822 -> Threshold:     4%
18:51:58.822 -> Compensation:  0x97
18:51:58.822 -> 
18:52:00.831 -> Device Reading:
18:52:00.831 -> Address:       0x32
18:52:00.831 -> Version:       3
18:52:00.831 -> ADC:           0
18:52:00.831 -> Voltage:       0.00 mV
18:52:00.831 -> Percent:       1.41%
18:52:00.831 -> Is Sleeping:   No
18:52:00.831 -> Alert:         Yes
18:52:00.831 -> Threshold:     4%
18:52:00.831 -> Compensation:  0x97
18:52:00.831 -> 
18:52:02.840 -> Device Reading:
18:52:02.840 -> Address:       0x32
18:52:02.840 -> Version:       3
18:52:02.840 -> ADC:           0
18:52:02.840 -> Voltage:       0.00 mV
18:52:02.840 -> Percent:       1.00%
18:52:02.840 -> Is Sleeping:   No
18:52:02.840 -> Alert:         Yes
18:52:02.840 -> Threshold:     4%
18:52:02.840 -> Compensation:  0x97
18:52:02.840 -> 
18:52:04.848 -> Device Reading:
18:52:04.848 -> Address:       0x32
18:52:04.848 -> Version:       3
18:52:04.848 -> ADC:           0
18:52:04.848 -> Voltage:       0.00 mV
18:52:04.848 -> Percent:       0.59%
18:52:04.848 -> Is Sleeping:   No
18:52:04.848 -> Alert:         Yes
18:52:04.848 -> Threshold:     4%
18:52:04.848 -> Compensation:  0x97
18:52:04.848 -> 
18:52:06.838 -> Device Reading:
18:52:06.838 -> Address:       0x32
18:52:06.838 -> Version:       3
18:52:06.838 -> ADC:           0
18:52:06.838 -> Voltage:       0.00 mV
18:52:06.885 -> Percent:       0.17%
18:52:06.885 -> Is Sleeping:   No
18:52:06.885 -> Alert:         Yes
18:52:06.885 -> Threshold:     4%
18:52:06.885 -> Compensation:  0x97
18:52:06.885 ->

porrey, 2022-03-28:
Since it runs on address 0x32 it is a cloned chip (not produced by Maxim Integrated). No are clones are bad, but most do not work. I have seen a few reports off good clones. For example, some sold on Ali Express seems to work. I purchased a clone from Amazon and could not get it to work. I had similar results as what you are seeing on yours. The one I purchased from SparkFun works flawlessly. I can't say for sure you have a non-functioning clone, but it is a possibility.

Here is a popular thread with some users going back and forth on ways to get them working.

https://forum.arduino.cc/t/max17043-fuel-gauge-not-working-at-all/329534/12

tmargot, 2022-03-29:
That is disappointing, I bought three of these from AliExpress. Oh well, as in UK Sparkfun themselves are out of the question, so will head over to Amazon.co.uk and hope the Sparkfun I buy there is an original. Much appreciate the help, thank you, and thank you for this library, it is brilliant work. Take care.

closed 2022-10-21 by porrey